terjeio/GRBL_MPG_DRO_BoosterPack #1: Congratulations!, great job!

luisluna2, 2018-09-27:
Hello friend just now I¿'m working in a similar project, but I wenna usa  a SD card but not have success with this. 

I make  a code with LCD 1602 I2C and now the machine can move but the problem is how send data from SD CARD, have some about this?

terjeio, 2018-09-27:
Thanks,

Funny you should mention SD card now, I just looked into how to hook code for streaming data from a SD card into my grbl port a few hours ago. So far I just added the hook, no code yet for interacting with the card.

 I have also added added code (not yet published) for streaming GCode via ethernet to the driver for the [EK-TM4C1294XL LaunchPad](https://github.com/terjeio/grblHAL/tree/master/drivers/TM4C129), this ment that I had to run grbl as a FreeRTOS task - in a similar vein as the ESP32 port found [here](https://github.com/bdring/Grbl_Esp32). The ESP32 port support for streaming from a SD card, maybe that implementation can give you some ideas?

Do you use a second microcontroller for your solution too?

luisluna2, 2018-09-27:
wow! good job!, could send some picture for undertand more?, sorry I'm new in Arduino hehe

terjeio, 2018-09-28:
@luisluna2 I have added some details about my setup to the readme that may be helpful. Implementing the setup you describe on a single microprocessor is IMO not going to be easy, the Arduino Nano surely does not have enough free resources for that. Perhaps the way forward will be to use your Mega as the Grbl controller, this with my patched version, and implement the DRO/MPG and SD card support on the Nano?

I do not work with Arduino controllers, my expertise is with TIs 16 bit controllers and ARM processors, so I can not offer much help for Arduinos. Maybe switching to the ESP32 I mentioned above is a better choice for your ambitions, but it still needs a second processor for some parts (such as keyboard input) as it has limited IO resources. Or you could adapt this project if you feel so inclined...