terjeio/GRBL_MPG_DRO_BoosterPack #3: Possible to buy an PIPico pcb?

mundsen, 2021-11-12:
Cool project

terjeio, 2021-11-13:
I have to order some PCBs first - been too busy to do that. I have one homemade prototype (single-sided) so hopefully the first run will be bug-free...

mundsen, 2021-11-13:
ok :-)

Have ordered a Tiva C , LCD and Pico

As a first step I hope to get the LCD working in combination with my grblHAL Pro

terjeio, 2021-11-13:
The Tiva C is not needed, the Pico is used as an alternative...

There is a MSP430 on the board that needs to be programmed, my plan is to port the programming code (it is public domain) so that the Pico can do that.
Currently I use a MSP-EXP430G2 to program it, it is possible that the Tiva C can be used as well. Anyway, I believe can supply a preprogrammed MSP430 - IIRC I have an adaptor for doing that off PCB.

terjeio, 2022-01-05:
FYI I have now received a few PCBs and have the first up and running with basic functionality in place.
I need to make a 5x5 keypad to assign and check out new key mappings required as the board only support a single MPG encoder so there is a little bit more work to do before I can release the code.

You still want a PCB? Unpopulated?

mundsen, 2022-01-05:
Yes 😊

Would like to test it with my laser project based on the grblHAL Teensy 4.1 Pro controller [Image.jpeg]

Last ned Outlook for iOS<https://aka.ms/o0ukef>

mundsen, 2022-01-08:
Is Grbl v1.1 jogging supported by grblHAL ?

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging

terjeio, 2022-01-08:
> Is Grbl v1.1 jogging supported by grblHAL ?

Yes, and I have fixed the bug where a jog cancel is sometimes not handled as it should.

Also, legacy Grbl errors out if a jog command exceeds max travel if soft limits is enabled - contrary to what the linked document claims. grblHAL has separate soft limits handling for jog commands that can be enabled by setting $40 to 1.

terjeio, 2022-01-13:
Testing is going well and I have ordered an adaptor for programming the MSP430 off board. Should be here soon.

Here is the BOM as extracted from KiCad:

[RP2040 MPG and DRO.zip](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack/files/7860577/RP2040.MPG.and.DRO.zip)

R1-9 has no value due to LEDs often having different current requirements for outputting the same brigthness. I use 2K7 for the 3mm LEDs in my test setup and the different colors vary from quite dim to very bright...
R1-9 could be replaced with 0R resistors on the board and having leaded resistors in series with the LEDs instead.

U3 is not needed if the supply voltage is 5V or the Pico is connected via USB. Short pin 1 and 3 with a wire if so.

DB15 is optional, use the connector(s) of your choice and wire to the pin headers as needed. I believe that wireless operation via HC-05 bluetooth modules is possble but I have not tested that yet.

The display is a [2.4" ILI9341 based panel](https://www.ebay.co.uk/sch/i.html?_from=R40&_trksid=p2334524.m570.l1313&_nkw=ili9341&_sacat=0&LH_TitleDesc=0&_odkw=ili9340&_osacat=0), I may add touch support later but not sure if it is going to be practical.

My test setup has all 25 buttons wired up, do you plan to add any? ([You will need at least two](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack/discussions/6#discussion-3786872)).

[image]

mundsen, 2022-01-13:
Possible to use something like this?

https://www.aliexpress.com/item/32635207928.html?spm=a2g0o.detail.1000060.1.6d4733dc9chmPd&gps-id=pcDetailBottomMoreThisSeller&scm=1007.13339.169870.0&scm_id=1007.13339.169870.0&scm-url=1007.13339.169870.0&pvid=af4624f1-85fd-40f5-8538-35733463f9a3&_t=gps-id:pcDetailBottomMoreThisSeller,scm-url:1007.13339.169870.0,pvid:af4624f1-85fd-40f5-8538-35733463f9a3,tpp_buckets:668%232846%238111%231996&&pdp_ext_f=%7B%22sceneId%22:%223339%22,%22sku_id%22:%2210000006002568188%22%7D

less wires - Seems like different keys gives different ohm values?

terjeio, 2022-01-13:
> Possible to use something like this?

Yes, but you will have to write code for it. Polling an AD-converter and decoding the value. You can ditch the MSP430 if doing that - and luckily the I2C SDA/SDC going to it are broken out to a pin header. Depending on how many key functions you require it may be a bit restrictive though - and I am not sure you will be able to get two key rollover to work.

mundsen, 2022-01-13:
Ok only an idea - I guess it’s best start with the same setup you use

Last ned Outlook for iOS<https://aka.ms/o0ukef>

mundsen, 2022-01-20:
Does/can the LCD show other live status information when not in MPG mode ?
Job %
Alarm messages
Use some of the buttons for functions like CycleStart, Feed Hold..?

terjeio, 2022-01-20:
> Does/can the LCD show other live status information when not in MPG mode ?
Job %
Alarm messages
Use some of the buttons for functions like CycleStart, Feed Hold..?

I have not (yet?) implemented Job % (only available when running code from SD card), the rest - yes.

carneeki, 2022-01-20:
> > Does/can the LCD show other live status information when not in MPG mode ?
> > Job %
> > Alarm messages
> > Use some of the buttons for functions like CycleStart, Feed Hold..?
> 
> I have not (yet?) implemented Job % (only available when running code from SD card), the rest - yes.

Is it running from the grbl SD card? If not, is there enough room to show current, and last couple of lines executed?

mundsen, 2022-01-20:
I have the PI, LCD..
Have to buy som eresistors..
So almost ready for the pcb :-)

terjeio, 2022-01-20:
> Is it running from the grbl SD card? If not, is there enough room to show current, and last couple of lines executed?

Currently any SD card has to be connected to the controller, not to the MPG. To display lines in the MPG they have to be transferred somehow. That would require code changes in the controller. It would be easier to add a SD card interface to the MPG board?

> So almost ready for the pcb :-)

And I am waiting for the TSSOP programming adapter, should be here any day now.

terjeio, 2022-01-29:
The adapter is here and I have a PCB and a preprogrammed MSP430 ready. Email terjeio@online.no for exchanging details.

mundsen, 2022-01-30:
Sent e-mail :-)

mundsen, 2022-03-22:
All parts arrived + a hot air soldering sation..

Hope to get some spare time soon - so I can get this pcb ready