terjeio/GRBL_MPG_DRO_BoosterPack #5: PiPico Board screw hole

carneeki, 2022-01-05:
Hi Terje,

In the PCB layout the J7 (controller) connector obscures a 3mm mounting hole (see first and second picture)
[image]

[image]

By rotating the VSUP header, and moving U3 further to the right, room can be made for both the screw hole and J7 (see third picture)
[image]

Also suggest using a 3.2mm hole; as it permits a #10-32 screw for those in the US.

I am hesitant to send a PR as I upgraded to KiCAD 6 and the files are not backward compatible (only forward compatible).

terjeio, 2022-01-07:
It is likely that I will change the D socket to a smaller version or replacing it with different connector(s). The latest version of the firmware I am working on can work with only four wires, or soon maybe even two (for power, with wireless transmission of data) or none (if battery operated).

What is more interesting is a discussion about how to connect to the controller and which kind of enclosures could be used. From that a more appropriate PCB design can be made.

FYI I made a 5x5 keypad yesterday for testing, and I have more than 30 commands assigned already. It seems like this can be a powerful/flexible platform to build upon...

carneeki, 2022-01-07:
Funny you mention enclosures, I am in the middle of designing one now that a friend offered to 3d print for me. My constraints are a 180x180 bed area.

[image]

I admit to going overkill with the screws, but that was an aesthetic choice ;)

[image]

The connector will likely be a CX20 or CX24 style connectors with a short pigtail from connector to board. (if you opt for 0.1" headers, that would permit a wide variety of connection options, including many cable to board connectors). CX24 permits up to 26 pins in and out of the enclosure (3A per pin) and I have found them quite cheap on AliExpress. Currently thinking:

- 15 pins for the MPG/DRO board to main controlbox (currently Pi3 + Grbl, stepper drivers, 5V, 12V and 48V PSUs),
- 2 pins for 5V/12V power and I have a number of spare lines I can use in the future.
- 9 pins free for future use

I haven't put a display or buttons in yet; one idea that I thought about was to use buttons next to the display similar to what many oscilloscopes offer. The buttons can be assigned different tasks depending on the screen you are looking at. Here is a spectrum analyser I've used, the 5 buttons below the screen are redefineable "soft" buttons:

[image]

Fluke 289 multimeter does this too:

[image]

Having, a few buttons at the bottom and the sides would open up many options for users interacting with the device. As I'm not sure what sort of displays you are planning to use, I was actually in the middle of looking your different repos to see if I could figure it out when you replied :)

My enclosure is probably far bigger than what most MPG pendants opt for, but something like this enclosure might suit the 5x5 keypad and the MPG wheel, with room for a few soft buttons too:

https://www.thingiverse.com/thing:2218467

mundsen, 2022-05-15:
I'm going to start on mine (solder the smd components on the pcb..  Anyone have one up and running that want to share images.. for inspiration?

mundsen, 2022-05-15:
> It is likely that I will change the D socket to a smaller version or replacing it with different connector(s). The latest version of the firmware I am working on can work with only four wires, or [...]

in the grblhal2000 Jog2K pendant project, they use an USB-C fron the pendant to the controller

terjeio, 2022-05-17:
> in the grblhal2000 Jog2K pendant project, they use an USB-C fron the pendant to the controller

If I am not mistaken a Pi Zero acts as a USB host to UART comms translator between the controller and the pendant.

andrewmarles, 2022-05-17:
> r

No, it is more simple than that.  The Jog2K uses a just a Pi Pico to emulate an I2C keypad to GRBLHAL.  The USB-C connector is just a physical interface that is convenient because it is compact, surface mounted and has enough conductors for KPSTR and e-stop as well as differential I2C (in the original implementation).  All of the communication is I2C, including the addition of DRO and status display.

I have not created a PR for the keypad code as it is kind of its own thing for the jog2k and Hal2K since the hardware is slightly bespoke.  But I am happy to formwat it as such if you are interested to take a look  Also I have plans to find a way to run a modified version your own MPG/DRO firmware on the jogger instead of my current code as I would prefer that the two projects converge rather than diverge, but sadly time is not infinite so I have not gotten to that yet.

andrewmarles, 2022-05-17:
Having said all that, I went ahead and created the PR in case it is useful.