terjeio/GRBL_MPG_DRO_BoosterPack #9: MSP-EXP432P401R serial port not working

CHTUZKI, 2022-09-22:
HI.
I want to try this project. I bought a MSP-EXP432P401R board and compiled the project. But the serial port of my board doesn't seem to work, I can't communicate with my MSP-EXP432P401R board using iosender, I use the virtual serial port of the onboard XDS110. I don't know what the problem is, can you help me

terjeio, 2022-09-22:
> I can't communicate with my MSP-EXP432P401R board using iosender

The firmware is for communication with a grblHAL controller with MPG mode enabled, you cannot connect ioSender to it...

[PB0/PB1](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack/blob/master/tivac_pins.md) is used for connecting to the CNC controller MPG UART port.

CHTUZKI, 2022-09-23:
Sorry, there is something wrong with my question.
I now use the MSP-EXP432P401R board to run the GrblHAL firmware and then the Tiva board to run the GRBL_MPG_DRO_BoosterPack firmware. I think that the iosender communicates with the first serial port of MSP-EXP432P401R, and the Tiva board communicates with the second serial port of MSP-EXP432P401R. Is this the case? If it is my iosender can not communicate with MSP-EXP432P401R.

terjeio, 2022-09-23:
Ok, then you are using the boosterpack map and have no EEPROM connected? If so set [these two symbols](https://github.com/grblHAL/MSP432P401R/blob/063abf7af633c9159a36b34d10b5c7ca2bd86276/cnc_boosterpack_map.h#L50-L51) to to 0 as the controller will hang if no EEPROM is present.

CHTUZKI, 2022-09-23:
I'm using the MSP-EXP432P401R board, which doesn't have any external expansion boards connected.

Below is my setup:
```
#define MPG_ENABLE          1
#define KEYPAD_ENABLE      0

#define CNC_BOOSTERPACK  1
#define I2C_ENABLE                0
#define EEPROM_ENABLE       0
```

But the problem still exists. iosender cannot communicate with grblhal through the virtual serial port of XDS110.

terjeio, 2022-09-23:
Even after changing the symbols?

CHTUZKI, 2022-09-23:
yes

CHTUZKI, 2022-09-23:
I checked the pins of the board with a multimeter and there is no problem

terjeio, 2022-09-23:
Ok, if you have downloaded the code with CCS can you stop execution with the pause button? Where does it hang?

And have you removed the jumpers connecting RXD and TXD between the debugger part of the board and the MCU part? You should not.

CHTUZKI, 2022-09-23:
I feel like the program is running normally, and I'm stepping through it. I put a breakpoint in the serial port interrupt function and I open iosn
`void SERIAL_IRQHandler (void)`
I feel that the serial port has no way to receive and send normally。
I'm going to compile an example now to verify that my board's serial port is OK.

terjeio, 2022-09-23:
I am running this driver with a CNC Boosterpack in my physical test machine, it has a 2K EEPROM and is running flawlessy. I just hooked up a spare board without anything connected (and EEPROM support disabled in the board file) and this starts too, albeit in Alarm mode since no pins are connected.

Note that this driver does not have support for saving settings in flash, either connect an EEPROM or hard configure it by uncommenting and editing definitions in _grbl/config.h_. Settings _can_ be changed at run time but will be lost on a hard reset.

CHTUZKI, 2022-09-23:
Thank you for your reply. I don't feel anything wrong with grblHAL now. Although my board still doesn't work. I'm not using CCS IDE, I'm using PlatformIO IDE, I found it might be the IDE's problem. I put a breakpoint on serialWriteS and in debug mode it prints out the characters, if I exit debug mode there is a problem.
I have had this problem many times with platformio, not only with MSP432, but also with STM32. It is possible to run step by step in debug mode, but not to exit debug mode. I will report the problem to platformIO.
I can't use CCS IDE because my chip may be pirated, and I can't upload programs using CCS IDE. I will buy another chip and retest.
Thank you very much~

CHTUZKI, 2022-09-25:
I replaced a brand new msp432p401r chip and downloaded it using CCS IDE without any problem. Everything works fine now.
I think it should be a problem with pirated chips or a problem with platformIO.

Thank you very much for your help. Thank you also for contributing such a cool project.

closed 2022-09-25 by CHTUZKI