terjeio/GRBL_MPG_DRO_BoosterPack #11: Can't run this project

CHTUZKI, 2023-02-01:
I want to run this project, but I have a lot of problems。

I use TM4C123GXL board as MPG controller，I use EXP432P401R board as grblhal controller,The source code of the grblhal controller is https://github.com/grblHAL/MSP432P401R.

I found that my MSP432P401R can't communicate with TM4C123GXL, I use a logic analyzer to measure the P3.2 and P3.1 pins of MSP432P401R, they don't have any data.（I use IOsender to communicate with MSP432P401R, and then use a logic analyzer to measure P3.2 and P3.1 pins，I think PC will give G code to MSP432, then MSP432 will use P3.2 and P3.1 pins to TM4C123. I don't know if I understand correctly）

My MSP432 board does not use EEPROM, I saw that this board can communicate with IOsender。

I debug MSP432 board and I found that
`if((hal.driver_cap.mpg_mode = stream_mpg_register(stream_open_instance(MPG_STREAM, 115200, NULL), false, NULL)))
        protocol_enqueue_rt_command(mpg_enable);`

will not execute。

terjeio, 2023-02-01:
> My MSP432 board does not use EEPROM

Have you changed the value in these lines in the map file from 1 to 0 to disable the EEPROM? If not the controller may hang.

```
#define I2C_ENABLE       1 // Only change if BoosterPack does not have EEPROM mounted
#define EEPROM_ENABLE    1 // Only change if BoosterPack does not have EEPROM mounted
```

When I try to compile the grblHAL driver I get a linker failure, solved by adding:

```
#if KEYPAD_ENABLE
#include "keypad/keypad.h"
#endif

```
after the EEPROM enable include in _driver.c_ (near line 45). 

>Does MPS432 have portC port? (GPIOC? GPIOC2?)

Yes, the ports can be referenced both by numbers or letters.

CHTUZKI, 2023-02-01:
> Have you changed the value in these lines in the map file from 1 to 0 to disable the EEPROM? If not the controller may hang.
>
> ```
> #define I2C_ENABLE       1 // Only change if BoosterPack does not [...]

CHTUZKI, 2023-02-01:
`#define CNC_BOOSTERPACK`
`//#define I2C_ENABLE`
`//#define EEPROM_ENABLE`
`#define MPG_ENABLE 1`

Here is my configuration。

I don't think my MSP432 board is hanging. I am iosender and can communicate with MSP432.

I don't understand why the P3.2 and P3.1 pins of the MSP432 board don't have any data output。

closed 2023-02-01 by CHTUZKI

terjeio, 2023-02-01:
Another "bug"...

Add  `serialRegisterStreams();` at line 1630 or so in driver.c (just before the driver cappabilities comment). At least this gives me serial transmit interrupts on Serial2...

CHTUZKI, 2023-02-02:
After adding the `serialRegisterStreams();` line, the system works normally。

🎉🎉🎉🎉thank you very much