terjeio/GRBL_MPG_DRO_BoosterPack #12: Can this project be used to control the A axis of grbl.

CHTUZKI, 2023-02-05:
I want to control my A-axis with this project, I don't know if it is possible with this project. If not, is it feasible for me to treat the X-axis signal as the A-axis signal. In grbl, are the six axes of XYZABC implemented with the same code?

terjeio, 2023-02-06:
> I want to control my A-axis with this project, I don't know if it is possible with this project.

With code changes, yes.

> If not, is it feasible for me to treat the X-axis signal as the A-axis signal.

Same as above.

> In grbl, are the six axes (XYZABC) implemented with the same code?

I am not sure what you mean by this. Additional axes (to XYZ) are implemented by extending data structures and replicating code for each. The replicated code is slightly changed to reference the new axis information. This project does the same, [here are some examples](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack/blob/56ae4170ad446d3c09dc4300b88698503fb00c35/Firmware/RP2040/src/grblcomm.c#L229-L284).

CHTUZKI, 2023-02-06:
Thank you for your answer

closed 2023-02-06 by CHTUZKI