terjeio/GRBL_MPG_DRO_BoosterPack #13: "GRBL_MPG_DRO_BoosterPack" project compilation error

CHTUZKI, 2023-02-05:
I download the project from this link。`https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack.git`

I use "Code Composer Studio 12.0.0" to open the `GRBL_MPG_DRO_BoosterPack-master\Firmware\TM4C123` project。I compile and the compiler tells me there are 16 errors。
`"../src/canvas/sdcard.c", line 177: error #256: type name is not allowed
"../src/canvas/sdcard.c", line 177: error #66: expected a ";"
"../src/canvas/sdcard.c", line 177: error #20: identifier "i" is undefined
3 errors detected in the compilation of "../src/canvas/sdcard.c".
gmake: *** [src/canvas/sdcard.obj] Error 1`

I made a modification to the code：

`if(!canvasSDCard) 
{
        canvasSDCard = UILibCanvasCreate(0, 0, 320, 240, handlerCanvas);
		listPrograms = UILibListCreate((Widget *)canvasSDCard, 10, 40, 300, LISTELEMENTS, handlerProgramList);
		{
		uint_fast8_t i;
		for(i = 0; i < LISTELEMENTS; i++)
			UILibListCreateElement(listPrograms, i, "", handlerSelectList);
		}
}`

error disappears。

CHTUZKI, 2023-02-06:
I made a modification to the code：
`__attribute__((weak)) void serial_RxCancel (void){};`

closed 2023-02-06 by CHTUZKI

terjeio, 2023-02-06:
Retested with Tiva C after RP2040 (Pi Pico) changes, seems ok now.