terjeio/GRBL_MPG_DRO_BoosterPack #15: How to use the keyboard in the "GRBL_MPG_DRO_BoosterPack" project

CHTUZKI, 2023-02-19:
Hi.

I want to use the keyboard, but I have encountered some problems. I use MSP430G2554 to drive the keyboard. The IIC interface of MSP430G2554 (keyboard board) is connected to tiva-c (DRO board), but the keyboard does not seem to work, and MPG cannot be used.

I don't know where to check. I looked at the code. The keyboard seems to be related to the UI. Can you give me some help?

Thank you.

terjeio, 2023-02-19:
Assuming you have buttons for jogging the first thing to check is the keypad strobe signal (PC4), is it going low while pressing a jog button?
Second thing is to check if the I2C bus is not stuck - both SDA (PA7) and SCL (PA6) should be 3.3V.
Do you have access to an oscilloscope or a logic analyzer?

CHTUZKI, 2023-02-20:
Thank you for your reply.

I have many questions about the keyboard processing of the "GRBL_MPG_DRO_BoosterPack" project. I am not very clear about this process
I found that MSP430G2554 is connected to the keyboard and communicates with tiva-c board through IIC, but MSP430G2554 is an IIC slave. If there is a key pressed, how can tiva-c know if there is a key pressed in time.

I found the following lines of code in the "GRBL_MPG_DRO_BoosterPack-master\Firmware\TM4C123\driver. h" file:

```
#define KEYINTR_PIN               GPIO_PIN_4          
#define KEYINTR_PORT            GPIO_PORTC_BASE     
```
What is this for？

I found a function in the "GRBL_MPG_DRO_BoosterPack-master\Firmware\TM4C123\i2c. c" file:

```
void i2c_nb_send (uint32_t i2cAddr, const uint8_t value)
{
    while(i2cIsBusy);

    i2c_m.buffer[0] = value;
    i2c_m.count     = 1;
    i2c_m.data      = i2c_m.buffer;
    i2c_m.state     = I2CState_AwaitCompletion;

    I2CMasterSlaveAddrSet(I2C1_BASE, i2cAddr, false);
    I2CMasterDataPut(I2C1_BASE, *i2c_m.data);
    I2CMasterControl(I2C1_BASE, I2C_MASTER_CMD_SINGLE_SEND);
}
```
I search for this function throughout the project. I found only two calls，

`leds_setState` and `keypad_setLeds`,

They just seem to get the status of LED, where to use `i2c_nb_send` function gets the keyboard data？
Can you give me some help? Thank you very much。

terjeio, 2023-02-20:
> where to use i2c_nb_send() function gets the keyboard data？

Data is read from the MSP430 in an [interrupt handler](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack/blob/f7d28290e91c365e3bd257cd5309562a7feeda15/Firmware/TM4C123/driver.c#L466-L489), the interrupt is triggered by the MSP430 pulling PC4 low. For most pins this is vith a short pulse, but for jogging the pin is held low until the key is released (key up event). When released the jog is terminated.

CHTUZKI, 2023-02-22:
[image]

closed 2023-02-22 by CHTUZKI

CHTUZKI, 2023-02-22:
I'm sorry. I accidentally turned off this issue.

I'm trying. But there are still problems.

I use MSP430G2554 to drive the keyboard, and I use "Code Composer Studio 12.0.0" to debug. It seems that everything is normal, but the I2C interrupt of MSP430G2554 will never enter. I have connected the pull-up resistance on P1.6 and P1.7, which is the data displayed by my logic analyzer.

[image]

CHTUZKI, 2023-02-22:
yes. When P2.5 is low level, tivaC will actively request data from MSP430G2554. At this time, MSP430G2554 should enter the iic interrupt, but it did not enter. I plan to migrate MSP430G2554 code to STM32.

CHTUZKI, 2023-02-23:
Whether the "GRBL_MPG_DRO_BoosterPack" project can independently control grblHAL? I want to make an offline controller to control the grblHAL board without using the "IOsender" software. Can this project be achieved

terjeio, 2023-02-23:
Can you zip your MSP430 binary and add it to a comment so I can test it here?

> Whether the "GRBL_MPG_DRO_BoosterPack" project can independently control grblHAL?

Yes, but it will need the gcode to be stored on a SD card in the controller as the MPG do not have a local SD card.
Without a SD card manual control is possible and was the reason for why I made it in the first place.

I'll look at updating the Pico version to support a local SD card - and also add support for Pico W to allow wifi access to the card via ftp or WebDav.

CHTUZKI, 2023-02-23:
Certainly. Thank you very much for your help.

There are some Chinese code comments in it, because I want to understand the running process of the code so that I can migrate to STM32. But it's still difficult for me, because I don't know what kind of data the program outputs. I hope code comments won't cause you trouble.
[KeypadProcessor.zip](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack/files/10811076/KeypadProcessor.zip)

terjeio, 2023-02-23:
> I don't know what kind of data the program outputs

It outputs a single character based on the key pressed - or rather makes it available for reading via I2C. There are several mapping tables - from keypad scan codes to the character they represents. [This is the default map](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack/blob/f7d28290e91c365e3bd257cd5309562a7feeda15/Firmware/Keypad%20processor/main.c#L188-L227).

CHTUZKI, 2023-02-23:
I tried to use arduino to simulate MSP430G2553. I manually triggered the `KEYINTR_PIN` pin of Tiva-C to let the arduino slave send a character, such as "U" and "F", but Tiva-C did not respond. I debug TivaC and found that TivaC could not trigger the `EVENT_KEYDOWN` event.

terjeio, 2023-02-23:
I have tested your version with the Launchpad connected directly to a STM32F446 grbrHAL controller with the keypad plugin enabled and it works, at least for the one button I tested - flood coolant toggle. This with a 2553 in the LaunchPad and a switch between P1.2 and P2.2. Are you testing with the LaunchPad connected to the Tiva?

> Did arduino send the wrong data,The program will use cnc_ map_ Default() function。

I cannot answer that - I do not have your Arduino code. And triggering the interrupt pin manually with a switch might be problematic due to switch bounce.

The logic analyzer trace above - was that triggered by a keypress? If so the Tiva attempted a read from they keypad processor but failed - could be due to incorrect wiring?

CHTUZKI, 2023-02-23:
Yes, I also suspect that there is a problem with my I2C connection. The picture of logic analyzer is triggered by cable short circuit. I use a wire to short P1.2 and P2.2,

CHTUZKI, 2023-02-23:
I will continue to test, thank you for your help

terjeio, 2023-02-23:
> I use 10K resistor as I2C pull-up resistor

Perhaps reduce that to 4K7 or 2K2. Anyway - the MSP430 should respond to a read request regardless of any key beeing pressed. [0 is returned if not](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack/blob/f7d28290e91c365e3bd257cd5309562a7feeda15/Firmware/Keypad%20processor/main.c#L648-L652). Or your MSP430 has blown pins?

CHTUZKI, 2023-02-23:
I re-ordered an MSP430G2553 from the Internet. I feel that there may be a problem with my chip. When the chip is mailed to me, I will continue to test it.

terjeio, 2023-02-24:
One thing to be aware of with the MPS430 LaunchPad is that P1.6 is connected to user LED2 via a jumper. IIRC this may interfere with I2C operation so perhaps it is good practice to remove the jumper? BTW it worked for me with the jumper in place but then I have pull-up resistors with a lower value than you have.

CHTUZKI, 2023-02-25:
There is no problem with the keyboard. happy. It's just a small problem. My I2C resistance may be too large. After I replace the resistance of 10K with the resistance of 4K7, the problem is solved.

It seems that my knowledge of hardware is still lacking😂

Thank you for your long help。