terjeio/GRBL_MPG_DRO_BoosterPack #16: MPG cannot be used

CHTUZKI, 2023-02-23:
Can the "GRBL_MPG_DRO_BoosterPack" project use MPG when the keyboard is unavailable?

My MPG is connected to the PC5 and PC6 pins of TivaC. When I rotate the MPG, nothing happens.

I connected a switch at the `MPG_MODE_PIN` pin and put the program in MPG mode. The X-axis label on the TFT screen also turned green.

CHTUZKI, 2023-02-23:
In the TM4C123/driver. c file:
```
static void MPG_X_IntHandler (void)
{
    uint32_t pos;
    static int32_t last = 0;

    QEIIntClear(QEI0_BASE, QEI_INTTIMER);

    if((pos = QEIPositionGet(QEI0_BASE)) != last) 
    {
        last = pos;
        mpg.x.position = ((int32_t)pos - MID_POS);
        mpg.x.velocity = QEIVelocityGet(QEI0_BASE);
        if(interface.on_mpgChanged)
        {
            interface.on_mpgChanged(mpg);
        }
    }
}
```
`interface.on_mpgChanged(mpg);`It will not be executed. I found that the function pointer will be cleared after the program enters the main interface.

In the file dro. c:
```
        case EventWidgetPainted:;
            uint_fast8_t i;
            char rpm[10];
            event = 0;
            isReady = false;
            setBackgroundColor(canvasMain->widget.bgColor);
            grbl_data = setGrblReceiveCallback(displayGrblData);
            for(i = 0; i < 3; i++) {
                if(axis[i].visible) {
                    axis[i].lblAxis->widget.fgColor = i == mpg_axis ? Green : White;
                    UILibLabelDisplay(axis[i].lblAxis, axis[i].label);
#ifdef LATHEMODE
                    if(i == X_AXIS)
                        displayXMode("?");
#endif
                }
            }
            drawString(font_freepixel_17x34, 5, RPMROW, "RPM:", false);
            sprintf(rpm, "%6.1f", mpg_rpm);
            lblRPM->widget.fgColor = Coral;
            UILibLabelDisplay(lblRPM, rpm);
            drawString(font_23x16, 220, RPMROW - 3, "Jog:", false);
            drawString(font_23x16, 87, STATUSROW, "Feed:", true);
            setJogModeChangedCallback(jogModeChanged);
            if(grbl_data->mpgMode) {
                displayMPGFactor(X_AXIS, axis[X_AXIS].mpg_idx);
                displayMPGFactor(Y_AXIS, axis[Y_AXIS].mpg_idx);
                displayMPGFactor(Z_AXIS, axis[Z_AXIS].mpg_idx);
            }
            mpgReset = true;
            leds = leds_getState();
            keypad_forward(!grbl_data->mpgMode);
            setKeyclickCallback2(keyEvent, false);
            NavigatorSetPosition(0, nav_midpos, false);
//            serial_putC(CMD_STATUS_REPORT_ALL); // Request realtime status from grbl
            active = true;
            break;
```

`setKeyclickCallback2(keyEvent, false);`Will clear `interface.on_ MpgChanged` function pointer.

terjeio, 2023-02-24:
> Can the "GRBL_MPG_DRO_BoosterPack" project use MPG when the keyboard is unavailable?

Not without modifying code or ensuring that the code starts with MPG mode on.

> My MPG is connected to the PC5 and PC6 pins of TivaC. When I rotate the MPG, nothing happens.

In DRO mode the MPG [encoders are polled](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack/blob/f7d28290e91c365e3bd257cd5309562a7feeda15/Firmware/TM4C123/src/canvas/dro.c#L196-L209) and the callback is set to NULL. Polling is only done when connected to a controller and [MPG mode](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack#mpg-microcontroller) has been accepted by it. MPG mode might be entered automatically on startup or requested by pulling the MPG mode pin low. The controller will respond with a `|MPG:1` element in the real-time report if accepted. The request is made when a keypress (0x0D - carriage return) is received from the keypad controller.
The MPG callback is only used in some of the data entry canvases to adjust values.

> I connected a switch at the MPG_MODE_PIN pin and put the program in MPG mode. 

Is the MPG connected to a controller that has MPG mode enabled? And is the MPG mode request accepted?
FYI [ioSender](https://github.com/terjeio/ioSender) greys out the UI if MPG mode is active.

CHTUZKI, 2023-02-24:
Powerful projects, I will continue to explore

closed 2023-02-27 by CHTUZKI