terjeio/GRBL_MPG_DRO_BoosterPack #17: Can the "GRBL_MPG_DRO_BoosterPack" project set the parameters of GrblHAL

CHTUZKI, 2023-02-27:
I see that grblHAL has many parameters and options. Can "GRBL_MPG_DRO" set grblHAL like iosender?

[image]

terjeio, 2023-02-27:
> Can "GRBL_MPG_DRO" set grblHAL like iosender?

No, not without a lot of work - canvases has to be added for editing the settings.

CHTUZKI, 2023-02-28:
If I add "Edit Settings Canvas" in the "GRBL_MPG_DRO" project

In order to meet the above requirements, does the grblHAL code need to be changed?

terjeio, 2023-02-28:
> In order to meet the above requirements, does the grblHAL code need to be changed?

No - when in MPG mode you have access to everything that a sender has.

CHTUZKI, 2023-02-28:
Thank you for your reply。

closed 2023-02-28 by CHTUZKI