terjeio/GRBL_MPG_DRO_BoosterPack #20: Pi Pico Board, LED resistor calculation?

Overlights, 2024-02-12:
Hello Terje,
Sorry to bother you.
I made some layout changes to your MPG DRO board, to make it a little more versatile (in my opinion), but the circuit is the same.
Among the various small modifications (I'll tell you about them in detail in another context), I added a double Pad to the LED resistors.
In this way, the predefined SMD resistors (on bottom) can be easily replaced with Custom resistors on Top
(1/4W, standard tht), simply cutting the track of the corresponding SMD resistor (the precise point is indicated on Bottom).

Finally, the calculation is needed for the correct resistance value, for a given LED Vf (based on the color chosen) and for a given current (say 10mA, as an average value).

I imagined it was:
R(led) = (Vcc - Vf) / Iled

So giving two examples:
(5 - 1.8) / 0.010 = 320 ohm (red LED, Vf=1.8V)
(5 - 3.6) / 0.010 = 140 ohm (white/blue LED, Vf=3.6V)

But in this post:
https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack/issues/3#issuecomment-1011856945
you said you used 2.7K resistors for standard 3 mm LEDs, obtaining different light intensities, I suppose depending on the color of the LEDs.

If the formula above was correct, with a res. from 2.7K, I would get about 1 mA for a red LED, 0.5 mA for a white one.
Something doesn't add up.
Where am I doing wrong?

Regards,
Marco

carneeki, 2024-02-12:
> #3 (comment)

Maybe it's meant to be 270 ohms? We can swap the resistance and I_led in
the equation to get I_led:

(5-1.8) / 270 = 0.012 = 12ma
(5-3.6) / 270 = 0.005 = 5ma

and don't seem unreasonable values, though 5ma might be too dim for some
use cases.

Overlights, 2024-02-12:
Hi Adam,
Exactly.
12 mA would be an absolutely reasonable value, while 5 mA, a bit low.
Having to choose a conservative standard value (for pre-soldered SMD resistors), it would be the one for an average LED current of 10 - 15 mA, for the lowest Vf value (red LED, approximately 1.8V).
So:
(5 - 1.8) / 0.010 = 320 ohms
(5 - 1.8) / 0.015 = 213 ohms
Or as you were saying, 270 ohms, for 12 mA.
All LEDs with higher Vf will have a proportionally lower brightness.

carneeki, 2024-02-12:
Hi Marco,

I'm assuming you're wanting to optimise either the number of resistors you
keep on the shelf (or maybe how many reels a pick'n'place machine will need
loaded). To that end, I'd suggest using the same value resistor either in
series or parallel, and the E24 resistor values are really well optimised
for both cases -

For red (1.8V):
(5-1.8) / 300 = 10.7mA

For white / blue (3.6V), 2x 300 in parallel:
(5-3.6) / 150 = 9.3mA

This comes at the compromise of needing to use more components, but they're
the same value. It also permits (with a solder jumper and the parallel
pair) the option to have this as something the end user can choose if they
are plugging LEDs into a header.

Overlights, 2024-02-12:
No Adam,
I'm not going to change the default value of the SMD LED resistors, just make sure it was correct.
I guess Terje can clarify this.

As written above, I modified the layout of the Board with a double SMD - THT pad, for all the LED resistors.
The SMD resistors will all have the same value, suitable for the worst condition described above (for example, 220 ohm, for a red LED with a Vf of 1.8V for approximately 15 mA, or 270 ohm for approximately 12 mA)
But anyone will be able to implement LEDs of other colors, therefore with a specific Vf, replacing the default SMD resistor with a 1/4W THT resistor of the most suitable value.

terjeio, 2024-02-12:
Not easy this as modern LEDs seems to vary quite a bit in brightness depending on the color and perhaps the manufacturer as well?. E.g. a test board I made with SMD LEDs uses resistors ranging from 680E to 4K7. So having the option of adding a second resistor is IMO a good idea.

Overlights, 2024-02-14:
Thank you Terje,
I mainly wanted to make sure that the calculation for the LED resistors was correct, so that I could choose a standard value for all SMD resistors.
As described above, I would use 270 ohm, a suitable value for a standard Red LED (3 or 5 mm.). So a Vf of around 1.8V for 12 mA.
LEDs with different characteristics can still be implemented, using the second THT Pad for custom resistors.